zeroth-robotics/zeroth-bot #26: servo.get_positions() read positions but servo isn't connected

jingxiangmo, 2024-10-18:
[image]

clear cache?

lol

hatomist, 2024-10-22:
partially by design, the read multiple servos command reads the memory chunk that has the latest values for the servos. We have a scan command to figure out if some of the servos are offline;
not sure what's the better path here for error handling. In ideal world all of the servos are connected and we feed data to the inference engine/classical control, right now if a servo gets disconnected it will report last known position. we need to decide if that's the best way to error handle or not (if a servo is disconnected/malfunctioning do we want to continue executing policy or panic)

jingxiangmo, 2024-10-22:
yeah, i think we need to have testing initializing scripts to figure out if the servos are connected etc (error handling)

closed 2024-11-02 by jingxiangmo