zeroth-robotics/zeroth-bot #90: Something wrong with environmental issues and a value bug of run.py

BitBelief, 2025-02-07:
**wrong message:**
(humanoid-os) **user@user-System-Product-Name:~/zeroth-bot$** python experiments/run.py
pygame 2.6.1 (SDL 2.28.4, Python 3.12.8)
Hello from the pygame community. https://www.pygame.org/contribute.html
INFO:robot:Robot initializing...
INFO:robot:Checking connection to robot...
INFO:robot:Successfully pinged robot.
INFO:robot:Scanning servos...
Error during robot operation: <_InactiveRpcError of RPC that terminated with:
	status = StatusCode.UNIMPLEMENTED
	details = ""
	debug_error_string = "UNKNOWN:Error received from peer  {grpc_message:"", grpc_status:12, created_time:"2025-02-06T17:10:41.2711877+08:00"}"
>
Traceback (most recent call last):
  File "/home/user/zeroth-bot/experiments/run.py", line 478, in <module>
    main()
  File "/home/user/zeroth-bot/experiments/run.py", line 468, in main
    stop_event.set()
    ^^^^^^^^^^
UnboundLocalError: cannot access local variable 'stop_event' where it is not associated with a value

**anotherone(about video-stream service):**
(humanoid-os) <font color="#4E9A06"><b>user@user-System-Product-Name</b></font>:<font color="#3465A4"><b>~/zeroth-bot</b></font>$ openlch start-video-stream
An error occurred: <_InactiveRpcError of RPC that terminated with:
	status = StatusCode.UNIMPLEMENTED
	details = ""
	debug_error_string = "UNKNOWN:Error received from peer  {grpc_message:"", grpc_status:12, created_time:"2025-02-06T17:10:18.204871362+08:00"}"
>

mrking66, 2025-03-22:
hey,have you been solved this problem?I have the same problem,too.It seems to be a gRPC error, and I'm not sure if the server has implemented the method called by the client.

BitBelief, 2025-04-11:
I'm sorry to tell you that I still haven't found a solution.If I find a solution I will leave a message on it.

closed 2025-05-24 by jingxiangmo